dword1511/stm32-vserprog #1: Which stm32 should I use

relax79, 2016-02-05:
Hello, i am work notebook repair bussiness and i am work with bios most time. I have a genius g540 programmer and is so slow. I am decide to build your programmer but i see first this page"https://github.com/dword1511/serprog-stm32vcp"  and you say STM32F103C8T6 48pin in readme but mcu is different in the schematic pdf STM32F103Rxxx an i am buy STM32F103R8T6, it is 64pin. I saw later this version of the programmer but same confusing happen to me again , you say STM32F103C8T6 but in the schematic mcu is STM32F103C(B)T6 . I am good electronic skill but i do not understand modern digital mcu's so much :) . I am download also prebuilt binaries from here "http://dword1511.info/dword/projects/serprog-stm32vcp/prebuilt/" My question is can i use STM32F103R8T6 and if i use which hex or bin  and schematic correct, and also missing in the first schematic xtal (i know 8mhz and 2x15 or 22pf) and a lot of parts value, correct uart pin points  . If i not use then which mcu can i buy ?
"http://www.direnc.net/STM32F103CBT6-MCU-32BIT-CORTEX-M3-72MHZ-LQFP-48,PR-23142.html"
"http://www.direnc.net/STM32F103C8T6-MCU-32BIT-CORTEX-M3-72MHZ-LQFP-48,PR-23141.html"
"http://www.direnc.net/STM32F103C8T6-Mini-Development-Board-ARDUINO,PR-22302.html"
"http://www.direnc.net/ARM-Cortex-M3-STM32F103C8T6-STM32-development-board,PR-15818.html"
I am so sorry for my English, i know messed up the spech. Thanks.

dword1511, 2016-02-05:
Thanks for pointing the error out, I did not notice it for years!
The difference between STM32F103Cxxx is just different amount of Flash and RAM available. Since STM32F103C8T6 has the least amount, firmware built for it should also run on others.
If you have a board at hand, please send the schematics to me so I can write the board.h for it and compile the firmware you will need.

relax79, 2016-02-05:
I am not have a board, only have STM32F103R8T6 mcu "http://urun.gittigidiyor.com/ev-elektronigi/st-stm32f103r8t6-200855568" (I just bought a new). If you say buy a new mcu or board please chose for me in the links in my previus message then i am bought new one, not problem if it is working properly. Thanks,regards.

relax79, 2016-02-05:
I am confuse first place because you say "just with minor modifications in the source code to assign correct GPIO for USB D+ pullup and LEDs." I thought editing requires different mcu's.like c8t6, cbt6, r8t6.

dword1511, 2016-02-05:
If you do not have a board and need one, then just go for the cheapest STM32F103 board. Any MCU in the series will work without problem.

If you do have the board to solder the MCU you have just bought, please refer to https://github.com/dword1511/stm32-vserprog/blob/master/pcb/v2/sch.pdf for correct connection. R2/R4 in this schematic is 20ohm or so, PB2 and BOOT0 just need some pull down, C4/C5 are 15 or 22pF, C1/C2/C3 0.1uF, C6/C9 10uF, C7/C10 0.1uF. R6 is a fuse. LED can be omitted if you wish. Then just use "make BOARD=stm32-vserprog-v2" to compile the firmware.

STM32F103 series are pin-to-pin compatible when the package is the same. When package is not same, the pinout is different, but firmware sees no difference in terms of pin name.

relax79, 2016-02-05:
I have to decide which one to buy then i am prepare everything, If i happen some trouble i will let you know.I am really thanks your kind support,answers and effort to build this good project.

closed 2016-06-09 by dword1511