dword1511/stm32-vserprog #2: Compile error when building for stm32-tiny042-v1

graugans, 2016-02-13:
There is an issue when building for stm32-tiny042-v1 or any stm32f0x boards

```
$ make BOARD=stm32-tiny042-v1
ln -sfT "boards/stm32-tiny042-v1.h" board.h
arm-none-eabi-gcc -O3 -Wall -g -fno-common -ffunction-sections -fdata-sections -DSTM32F0 -mthumb -mcpu=cortex-m0 -msoft-float -Ilibopencm3/include/  -c vserprog.c -o vserprog.o
In file included from vserprog.c:14:0:
vserprog.c: In function 'handle_command':
spi.h:11:42: error: 'GPIO_SPI1_NSS' undeclared (first use in this function)
 #define SPI_SELECT()   gpio_clear(GPIOA, GPIO_SPI1_NSS)
                                          ^
vserprog.c:193:7: note: in expansion of macro 'SPI_SELECT'
       SPI_SELECT();
       ^
spi.h:11:42: note: each undeclared identifier is reported only once for each function it appears in
 #define SPI_SELECT()   gpio_clear(GPIOA, GPIO_SPI1_NSS)
```

dword1511, 2016-02-13:
Hi, support for STM32F0x2 is not completed yet. I will try to finish it when I have time.

graugans, 2016-02-13:
Thx, this would be really cool

dword1511, 2016-03-03:
Implemented and tested, ready for use. STM32F0 version seems to be ~70% slower than STM32F1 version. Whether it is due to clock or configuration is still unknown.

closed 2016-03-03 by dword1511