dword1511/stm32-vserprog #12: 4-byte addressing

eswierk, 2017-06-22:
It would be nice to support larger-capacity SPI NOR flash chips like W25256FV and N25Q256A that require 4-byte addressing.

dword1511, 2017-06-22:
`flashrom` currently does not support 4-byte addressing as far as I known. There were [patches](https://mail.coreboot.org/pipermail/flashrom/2015-January/013168.html) proposed but they are probably unmerged.

eswierk, 2017-06-22:
I've managed to get a flashrom with those patches working well enough to program an N25Q256A using an FT232H module, but it's awfully slow.

https://github.com/skyportsystems/flashrom/commits/master

eswierk, 2017-07-03:
It is possible to program N25Q/MT25QL chips larger than 16 MB with the existing flashrom serprog, using only commands with 3-byte addresses, plus the extended address register for the 4th byte. See https://github.com/skyportsystems/flashrom/commit/db3ca240 for an example.

Other chips may require native 4-byte addressing, but I haven't encountered them so far.

dword1511, 2017-07-03:
With a proper patch for `flashrom`, any serprog should be able to handle 4-byte addressing. It is merely an SPI-to-USB bridge. It has no knowledge about the protocol. The commands with address parameters in serprog protocol are for parallel flash chips. The only limitation for serprog with SPI is that the size of each transaction (command or data) must be under 16MB.

eswierk, 2017-07-03:
Ah! Thanks for the clarification. I confirmed that programming an M25QL256 with flashrom serprog works with direct 4-byte address commands. See https://github.com/skyportsystems/flashrom/commit/d30fcd40 .

closed 2017-07-03 by eswierk