dword1511/stm32-vserprog #13: Build is failing on xenial

Ntemis, 2017-06-24:
make --directory=lib/stm32/f1 SRCLIBDIR="/media/demetris/c4487cfd-d71a-441b-b83d-23d8ce19475d/stm32-vserprog/libopencm3/lib"
make[3]: Entering directory '/media/demetris/c4487cfd-d71a-441b-b83d-23d8ce19475d/stm32-vserprog/libopencm3/lib/stm32/f1'
  CC      i2c_common_v1.c
arm-none-eabi-gcc -Os -Wall -Wextra -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -Wundef -Wshadow -I../../../include -fno-common -mcpu=cortex-m3  -mthumb -Wstrict-prototypes -ffunction-sections -fdata-sections -MD -DSTM32F1 -ggdb3 -O3 -Wall -g -fno-common -ffunction-sections -fdata-sections -funit-at-a-time -fgcse-sm -fgcse-las -fgcse-after-reload -funswitch-loops -DSTM32F1 -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Ilibopencm3/include/  -o i2c_common_v1.o -c ../common/i2c_common_v1.c
../common/i2c_common_v1.c: In function 'i2c_write7_v1':
../common/i2c_common_v1.c:485:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
  for (size_t i = 0; i < n; i++) {
  ^
../common/i2c_common_v1.c:485:2: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
../common/i2c_common_v1.c: In function 'i2c_read7_v1':
../common/i2c_common_v1.c:507:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
  for (size_t i = 0; i < n; ++i) {
  ^
../../Makefile.include:44: recipe for target 'i2c_common_v1.o' failed
make[3]: *** [i2c_common_v1.o] Error 1
make[3]: Leaving directory '/media/demetris/c4487cfd-d71a-441b-b83d-23d8ce19475d/stm32-vserprog/libopencm3/lib/stm32/f1'
Makefile:63: recipe for target 'lib/stm32/f1' failed
make[2]: *** [lib/stm32/f1] Error 2
make[2]: Leaving directory '/media/demetris/c4487cfd-d71a-441b-b83d-23d8ce19475d/stm32-vserprog/libopencm3'
Makefile:102: recipe for target 'libopencm3/lib/libopencm3_stm32f1.a' failed
make[1]: *** [libopencm3/lib/libopencm3_stm32f1.a] Error 2
make[1]: Leaving directory '/media/demetris/c4487cfd-d71a-441b-b83d-23d8ce19475d/stm32-vserprog'
Makefile:51: recipe for target 'all' failed
make: *** [all] Error 2

dword1511, 2017-06-25:
Please try `make EXTRA_CFLAGS="-std=gnu99"`. Also please post the output of `arm-none-eabi-gcc -v`.

dword1511, 2017-06-26:
Confirmed that `make EXTRA_CFLAGS="-std=gnu99"` worked on Xenial. Now included in Makefile.

closed 2017-06-26 by dword1511