dword1511/stm32-vserprog #15: ~~x�~hellowor problem

atesonur, 2017-11-10:
Thank you so much for your code, it helped me a lot. But I have a problem. When I start up my stm32f103 device and try to read 14 byte from virtual port,  First 6 bytes are  always like ~~x�~ these.
Do you have any suggestion and thanks a lot for the code you support.

dword1511, 2017-11-10:
Check whether there are other programs accessing the serial port.

atesonur, 2017-11-10:
Thank you so much for your time. None of my programs are using serial port and even I restarted my computer just in case. I am using linux and after restart,  I got these:

master@Lenovo:~/Desktop$ sudo python echo2.py helloworldonur
~~x�~hellowor
master@Lenovo:~/Desktop$ sudo python echo2.py helloworldonur
ldonurhellowor

Thank you so much.

dword1511, 2017-11-10:
I do not know what you are using. Can you use `minicom`?

Make sure `ModemManager` is not running.

atesonur, 2017-11-11:
Thank you so much. I used minicom. It is same but i used a similar command handler like you do. It works fine. Thanks a lot for your time.

dword1511, 2017-11-11:
Without the exact description of your code and what you are doing, I cannot help you. If it is a project of your own, please use email rather than issues. Thanks!

closed 2017-11-11 by dword1511