dword1511/stm32-vserprog #43: Unable to get past make BOARD=stm32-bluepill

ginL, 2020-07-31:
It seems I cannot get past this. Tried POP_OS and mint. 

Here is a copy of the error code

```
mint@mint:~/stm32-vserprog$ sudo make BOARD=stm32-bluepill
First run, cleaning...
make[1]: Entering directory '/home/mint/stm32-vserprog'
rm -f vserprog.o usbcdc.o spi.o README.html stm32-vserprog.elf stm32-vserprog.hex stm32-vserprog.bin stm32-vserprog.map stm32-vserprog.out board.h last_board.mk
make[1]: Leaving directory '/home/mint/stm32-vserprog'
make[1]: Entering directory '/home/mint/stm32-vserprog'
git submodule init
git submodule update --init
CFLAGS="-O3 -Wall -g3 -gdwarf -std=gnu99 -fno-common -ffunction-sections -fdata-sections -funit-at-a-time -fgcse-sm -fgcse-las -fgcse-after-reload -funswitch-loops -DSTM32F1 -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -Ilibopencm3/include/ " make -C libopencm3 lib/stm32/f1 PREFIX=arm-none-eabi- V=1
make[2]: Entering directory '/home/mint/stm32-vserprog/libopencm3'
  GENHDR  include/libopencm3/stm32/f0/irq.json
./scripts/irq2nvic_h ./include/libopencm3/stm32/f0/irq.json;
/usr/bin/env: ‘python’: No such file or directory
make[2]: *** [Makefile:59: include/libopencm3/stm32/f0/irq.json.genhdr] Error 127
make[2]: Leaving directory '/home/mint/stm32-vserprog/libopencm3'
make[1]: *** [Makefile:116: libopencm3/lib/libopencm3_stm32f1.a] Error 2
make[1]: Leaving directory '/home/mint/stm32-vserprog'
make: *** [Makefile:64: all] Error 2
mint@mint:~/stm32-vserprog$ 

```

dword1511, 2020-08-01:
You need to install python.

Not sure if libopencm3 uses python3 now, will check.

dword1511, 2020-08-25:
libopencm3 is still using python2 as of now. Some distro no longer comes with python2 by default, you will need to install it manually.

closed 2020-08-25 by dword1511