dword1511/stm32-vserprog #45: 几个问题，望回复

jiuweiljp, 2020-08-25:
花了10.8买了一个stm 的demo板
原理图：
http://lzdz.f3322.net:81/serprog-stm32vcp/STM32F103C8T6最小系统原理图.pdf
照片：
http://lzdz.f3322.net:81/serprog-stm32vcp/1.jpg
http://lzdz.f3322.net:81/serprog-stm32vcp/2.jpg
http://lzdz.f3322.net:81/serprog-stm32vcp/3.jpg
也不清楚这个stm32 F103C8T6的真假，反正我是看不出来。

（1）、编译使用：
make BOARD=stm32-vserprog-v2
make BOARD=stm32-vserprog-v3
生成的固件简单测试都能读写，不知道那个更匹配我的的这个板
USB D+(PA12)因这个板上有3.3v的接线端子，所以我直接焊了一个1.5k的1/4瓦的电阻上去了。

2、led的问题，修改配置
boards/stm32-vserprog-v2.h

/* org
#define BOARD_RCC_LED                RCC_GPIOA
#define BOARD_PORT_LED               GPIOA
#define BOARD_PIN_LED                GPIO0
*/
/* mod */
#define BOARD_RCC_LED                RCC_GPIOC
#define BOARD_PORT_LED               GPIOC
#define BOARD_PIN_LED                GPIO13
/* mod  end */
#define BOARD_LED_HIGH_IS_BUSY       true

然后连接在pc13上的led灯工作了
led状态我观察了下，有如下几种：
1、长亮（IDLE？）
2、频率大概一秒的闪烁（不知道什么状态）
3、读flash的时候快速闪烁（灭的占空比高）
4、擦除flash的时候快速闪烁（亮的占空比高）
5、写flash的时候基本不灭（仔细观察感觉亮的占空比更高）
望告知1-2的状态，另外不知道3-5的状态是否正确

（3）、我的这个demo板用了2个晶振，那个接PC14，PC15的晶振taobao的介绍说是爱普生的32768Hz的，不知道能不能利用它，让读写的速度更快些，谢谢

dword1511, 2020-08-25:
1. 两种固件的 USB pull-up 不一样。你用的这块板子没有 pull-up，我不清楚为何 USB 能正常工作 （STM32 内部的 pull-up 应该不够）。除此之外没有区别。V3 因为使用 PA13 作为 pull-up，SWD 会不可用。
2. 1 是 idle，2 应该是读取。具体跟 Flash 有关（站空比取决与每个命令所需的时间）。
3. 32k 晶振用于 RTC，这里用不到。

jiuweiljp, 2020-08-26:
感谢你抽出时间回复！

有没有办法在windows计算程序运行时间？

jiuweiljp, 2020-08-26:
我用万用表测量过，这个demo板在PA12与3.3V之间有电阻，阻值有4K多。
如果用20030331的固件，有时候插上USB会出现unknown的设备（win 7），但是刷stm32-vserprog项目编译的固件后没有发现上述现象。

jiuweiljp, 2020-08-31:
如果你什么也没干，IDLE灯如果闪烁（间隔大约1秒），哪就是你的usb 连接有问题，重新拔插一下吧。