dword1511/stm32-vserprog #50: 关于QSPI模式的支持

Droid-MAX, 2021-02-25:
最近在研究怎么使用QSPI模式来给器件编程，但网上案例的实现方式都不尽相同，比如qspi_loader这个项目，使用非serprog协议，上位机用python编写，除此之外还有max1000_flashrom这个项目，使用fpga方案实现。所以我想serprog协议能否支持QSPI模式？该从哪个角度入手？

Droid-MAX, 2021-02-25:
https://github.com/dword1511/stm32-vserprog/issues/12

dword1511, 2021-02-25:
QSPI 跟 4-byte address 是两回事。4-byte address 用于支持 32M 和更大的闪存。

QSPI 有四根数据线。大部分 STM32 不支持这种接口。新一点的 F4 有支持。不过意义不大，因为 USB 2.0 FS 跑不了那么快，除非加上 HS PHY 。

Droid-MAX, 2021-02-26:
那如果想解决4-byte地址模式这个问题的话是不是还是得从上位机入手？比如修改flashrom的源码？

dword1511, 2021-02-26:
最新的 flashrom 应该已经支持 4BA 了
https://www.flashrom.org/Flashrom/1.1

Droid-MAX, 2021-03-01:
我知道问题所在了！
对于Winbond W25Q256FV闪存芯片来说，flashrom确实已经支持4Byte地址模式操作，不过决定闪存工作在3Byte/4Byte地址模式的ADP寄存器状态并未被设置为4Byte地址模式，所以导致系统无法正常启动的问题。同时我也找到了一个针对W25Q256FV的ADP寄存器状态设置的修改版flashrom，不过还没测试，希望可以解决我这问题。
https://github.com/huming2207/flashrom_google
我也给编译到了Windows平台上
https://github.com/Droid-MAX/flashrom_mod/releases/tag/1.0

Droid-MAX, 2021-03-02:
已结案，故close该issue.

closed 2021-03-02 by Droid-MAX